VoronDesign/Voron-Tap #10: Fun Fact Voids™

Durahl, 2022-11-27:
The on Page 27 of the Manual mentioned `Fun Fact Voids™` didn't appear with my settings of **SuperSlicer** and since I couldn't figure out what settings was responsible for them not to make an appearance I went ahead and enlarged them to a 0.2mm gap which then sliced them as intended.

Maybe consider increasing them in the STLs like I did or give a mention in the Manual what Slicer setting may be responsible for them to appear ( or not )?

[image]

pl4nkt0n, 2022-11-28:
You can see them within the STEP:
[image]

I think, your slicer is closing this gaps because of too high "gap closing radius". There should be the same function with different names in different slicers.

notgwj, 2022-11-28:
The SuperSlicer setting that causes this is actually `hole_size_compensation` (`XY compensation`).  Setting it to zero makes the voids work as intended.  See https://discord.com/channels/460117602945990666/1041007230239125574/1046753665270165574

Badnoob-327, 2022-12-13:
"This part has thin voids inside it (blue).  When the part is sliced, these produce internal ribs which strengthen the part.  If your slicer closes these up, recheck your slicer settings for hole size compensation. Default voron slicer profiles are available for voron printers."

Hows that? I wanted to keep it general as there are other settings that could affect this.

notgwj, 2022-12-13:
Looks good to me.

nemgrea, 2022-12-27:
fixed via https://github.com/VoronDesign/Voron-Tap/pull/32

closed 2022-12-27 by nemgrea