VoronDesign/Voron-Tap #15: MGN12 Carriage does not lay flat on Tap_Center

joseph-greiner, 2022-11-28:
I was looking at the CAD for the Tap an noticed interference between the MGN12 carriage and the tap_center component. I printed the STL in the repo, mounted the carriage with the belts installed and noticed that the parts mount at a slight angle to each other. 

[image]

The issue is that the chamfer for the bottom of the carriage on the tap_center part is too big and will not let the carriage lay flat, even with belts installed.

[image]

By cutting out the large chamfer and adding a 0.4mm chamfer to the area the carriage now lays flat.

[image]

Modified STL and STEP files can be provided if requested.

Badnoob-327, 2022-11-28:
it appears that some MGN12's have a different chamfor spec than others, I will update the STL.

Badnoob-327, 2022-11-29:
> I was looking at the CAD for the Tap an noticed interference between the MGN12 carriage and the tap_center component. I printed the STL in the repo, mounted the carriage with the belts installed and [...]

please look at the updated tap_center_r2 file and let me know if you see need for more than that, I like to keep some chamfer there if possible

joseph-greiner, 2022-11-29:
I am printing it now, I will update when it is finished, but the support was not exported with the part STL.

[image]

joseph-greiner, 2022-11-29:
Printed the r2 STL, the carriage sits flat now.

[image]

Badnoob-327, 2022-12-06:
I think this can be closed now, thank you for your feedback.

closed 2022-12-06 by Badnoob-327