VoronDesign/Voron-Tap #19: Wrong link to "Tap-Klipper Instructions"

GuidoHendriks, 2022-12-01:
First things first; I'm so impressed with this project! Thanks a lot. 👍🏻 

---

Now the minor issue with the manual...

On page 37 of the manual it says:

> The software setup and other initial setup steps with your new printer can also be found on our documentation page. We recommend starting with the **Tap-Klipper Instructions** on our Github.

This contains a wrong link.

Have double checked that I have the latest file. Also noticed this https://github.com/VoronDesign/Voron-Tap/issues/14 which also does not seem to be fixed in the latest file. Maybe it's not committed or pushed to Github yet?

Badnoob-327, 2022-12-13:
got it , and I rerefixed the #14, appreciate the reminder.

nemgrea, 2022-12-27:
fixed via https://github.com/VoronDesign/Voron-Tap/pull/32

closed 2022-12-27 by nemgrea