VoronDesign/Voron-Tap #20: Assemble Sensor instructions should address position of solder work

ditdafivo, 2022-12-02:
I assembled my OBP991T11Z to basically look identical to the picture on the manual page 13, even to the same scale, but as a result there isn't enough space to route the wires with the shrink tube, soldering joints etc. in the wiring channel of Tap_Upper_Wired_r1.stl.

I'm thinking now the best place for that all the solder work is so that it lives along side all the rest of the consealed wiring for the tool head.  In fact for the second try, I might not even do that solder work until the sensor body is screwed into Tap_Upper_Wired and the wires are installed in the wiring channel. I'm not sure if that's what others are doing or if specific length guidelines can make sense to document in the manual given the variety of toolheads, canbus mods etc.  But at least it would be helpful to note on that manual page that the picture is not meant to show the extra length required between the sensor body and the resistor/solder work/shrink tube to route through the wiring channel.

ditdafivo, 2022-12-05:
Having now fitted my stealthburner/CW2, I'd leave at least 30mm wire untouched out of the sensor body to be able to conceal with work safely under the toolhead wiring cover.

kyleisah, 2022-12-13:
@ditdafivo hey there, I made the graphic you're referring to. It wasn't really *intended* to be used to scale, but I can see how that would happen, and I think I would feel better having the actual components being tucked inside the cable door with everything else. I've got the CAD I made, wired sensors, and gumption. Maybe I just overlooked that with all the excitement. 😅 I'll talk to the right people and see what we can do to improve that.

Badnoob-327, 2022-12-16:
the intent there is actually to have the resistor tucked into the wire channel, I'll clarify the manual

ditdafivo, 2022-12-17:
I don't see how fitting inside the wire channel could have worked for my build. Maybe I needed smaller resistors/heatshrink. Or maybe I overextruded.

nemgrea, 2022-12-27:
fixed via https://github.com/VoronDesign/Voron-Tap/pull/32

closed 2022-12-27 by nemgrea