VoronDesign/Voron-Tap #21: Holes for CW2 mounting not centered.

krankydonkey, 2022-12-05:
Upon reviewing the CAD, it appears that the holes for the CW2 Mounting on Tap Upper Wired part are off centre, causing them to be out of alignment with the CW2 holes when it is inserted into the slot.
[image]

[image]

Badnoob-327, 2022-12-06:
Hi, thank you for the question.  I would not use the sides as reference, as their location is determined by other geometry

I rechecked the CAD, and relative to the hotend mounting screws there is a slight offset showing of approx .08 mm. This should not cause any real issues with assembly or operation, and has not caused any issues I'm aware of.

gji, 2022-12-10:
I also noticed some more centering issues on the center body - it looks like the MGN9 rail inset is a bit less than 0.1mm off as well, from the center of the holes. Seeing as people have this working it probably doesn't matter so much functionally, but it does make any modifications a pain since the CAD is pretty far from symmetric (e.g. the belt and magnet holes are also not symmetric) so one has to be very careful when picking reference surfaces.

[image]
[image]

Badnoob-327, 2022-12-16:
I'm aware of it, thank you.