VoronDesign/Voron-Tap #24: Brass threaded insert causing magnet not to sit flush

contrapuntal, 2022-12-10:
A M3x5x4 threaded insert causes bulging in the magnet holder, so that the magnet won't sit flush. 
[image]

mrmultifunk, 2022-12-11:
I had this happening as well. What worked for me, was to reheat the insert and press the bulging down with an Allen key.

krankydonkey, 2022-12-12:
What I found worked was inserting the magnet first and then putting the heat set insert in while holding the magnet flush with the side with a ruler

Badnoob-327, 2022-12-13:
I think krankydonkey has a good idea, and we can fix that in documentation (by switching the order and doing magnets before heatets.

Badnoob-327, 2022-12-16:
Next manual has the order of operations switched (magnets in first, then inserts), so this should be fine

ps-bond, 2022-12-20:
> Next manual has the order of operations switched (magnets in first, then inserts), so this should be fine

Is there an issue with breaking the adhesive if that's done?  My usual way of releasing superglued parts is to warm them...

Badnoob-327, 2022-12-20:
I've not had any issues with the superglue getting loose here, if you wish you could try RTV instead of superglue.

ps-bond, 2022-12-20:
Actually I've been using Loctite 480...  I've been using the stuff for a while to glue magnets to prints and have found it more resilient than standard CA.  Probably overkill, admittedly.

nemgrea, 2022-12-27:
fixed via https://github.com/VoronDesign/Voron-Tap/pull/32

closed 2022-12-27 by nemgrea