VoronDesign/Voron-Tap #27: More information for [stepper_z] section in klipper firmware manual for Tap needed.

kyleisah, 2022-12-13:
Hey there! Small issue, but I think it should also be mentioned that you'll need to remove the `[stepper_z] position_endstop` that *may also* be present at the **bottom** of printer.cfg from previous configuration.

Thanks! ❤️

Badnoob-327, 2022-12-16:
sounds good, feel free to do a pr on that

hyperphonic0, 2022-12-16:
Duplicate of #26 (and it is already mentioned in the instructions: https://github.com/VoronDesign/Voron-Tap/blob/main/config/tap_klipper_instructions.md).

kyleisah, 2022-12-17:
@hyperphonic0 I think you may have misread the issue. Yes, the current instructions tell you to comment out `[stepper_z] position_endstop` but it really should also mention to remove it from the bottom of your config, like what happens when klipper does a `save_config` command to save your new endstop value from before installing Tap. It's there at the bottom where it can be easily forgotten or missed.

closed 2022-12-17 by kyleisah