VoronDesign/Voron-Tap #40: Consider increasing depth of solder blob relief in Tap_Upper_PCB CAD from 1mm to 2mm, to reflect actual pin length/needed relief

nearctx, 2022-12-27:
Currently the TAP_Upper_PCB_r2 part has a 1mm deep relief for the pins and solder joints of the JST connector on the OptoTap v2.1 PCB so the PCB can sit flush when screwed down, however the pins of the spec connector(S3B-PH-K) in actuality stick out 1.8mm from the bottom of the PCB. Extending the depth of the relief to 2mm should resolve this issue.

Of course it is not difficult to just trim the pins down a little before or after soldering, but extending the relief to 2mm is a small change that would make assembly easier, and eliminate the opportunity for issues arising from the PCB being installed incorrectly due to the pins blocking it from sitting flush.

The measurement of the pins protruding 1.8mm from the PCB bottom is taking the specified PCB-side pin length of the JST connector, 3.4mm(Page 2 of the JST PH connector datasheet - see below for screenshot and link), then subtracting the nominal PCB thickness of 1.6mm, = 1.8mm protrusion. So a 2mm relief should be enough to account for any potential variance in pin length/PCB thickness.
[image]
https://www.jst-mfg.com/product/pdf/eng/ePH.pdf

I have also personally experienced this issue with the solder relief being insufficient with the current 1mm depth, and I have physically measured the pin length protruding to indeed be about 1.8mm as well.

Also I'd like to take this opportunity to say thanks to everyone who has contributed to Tap, its awesome!

raymondh2, 2023-01-02:
@Badnoob-327 @tanaes

blownupp, 2023-02-26:
I'll second this - I had to trim the pins down a little bit to install the PCB, but it isn't critical either...