VoronDesign/Voron-Tap #45: PCB needs EMI and grounding improvements (and other remarks)

tionebrr, 2023-01-08:
I checked the project yesterday.

Here are some recommendations; those will improve the EM susceptibility and characteristics of the board:
- Reduce the ground planes margin to near the manufacturer capabilities (0.127mm for JLCPCB -> 0.15-0.2mm).
- Add vias to stitch the ground planes together
- Make the board such that all traces are routed inside a ground plane.
- EMC profile of a 2-layer board tends to improve when the board gets thinner. Avoid using the standard 1.6mm board.

Those are some ameliorations on the design:
- The copper layer should be retracted from under the screws. Humidity will creep under a damaged soldermask. That kind of details make boards work 20 years longer for free. ;)
- The AP63205 datasheet states that the 220nF C2 could be 100nF 0603. Not sure what's going on here, I may miss something, but if it works at 100nF, go for it. 
- The switching node of the AP63205 is very compact (L1 very close the the SW pin), which is perfect. I would use a small copper pour to connect this node it a bit better.

Some EDA tricks:
- Name your schematics nets. It's making the routing and the editing way easier.
- You don't need to create a plane for TOP and BOT copper in kicad, you can have a single plane pouring on multiple layers.
- Lock the positions of all mechanical components. I shouldn't be able to move the screws holes for example ;)

I've joined a zip file with some quick modifications to the design you can look at. I have some spare time if you need any help for design and testing. ;) 
[Tap_Photosensor_PCB_v2_modified.zip](https://github.com/VoronDesign/Voron-Tap/files/10367618/Tap_Photosensor_PCB_v2_modified.zip)

TorbanP, 2023-02-14:
I just incorporated your changes into a board that also integrates a TVS. I like the comment on copper layers under screws; I haven't heard that one before. The other changes were perfect (switching node) , although sticking a tvs was a bit tricky due to spacing. I just ordered 10 and will dial in the TVS
[image]

tionebrr, 2023-02-21:
Hello @TorbanP , that's looking way better already!
Closing ;)

closed 2023-02-21 by tionebrr