VoronDesign/Voron-Tap #48: Toolhead mounting screw clearance seems to be wrong

ressu, 2023-01-18:
Yes, a very very minor thing. I think in the manual, on page 25, it's stated that the toolhead holder screw needs a 2.5mm clearance. But considering that the mount on SB parts that slots into those screws is ~3.3mm wide the correct value should be 3.5mm.

With 3.5mm clearance the ends of the screws also seem to match how the part was designed and the ends of M3x16 screws don't dig into the plastic on `Tap_Upper_*` part.

[image]

claudermilk, 2023-01-23:
I ran into that as well. I set the screws to 2.5mm as the manual directed, but had to back them out a bit to get the SB hotend mount to fit.

Badnoob-327, 2023-01-25:
I'll take a look, thanks