VoronDesign/Voron-Tap #49: Upper/Front attachment causes large warp

GaryBoone, 2023-01-23:
On page 25 of the manual, the `upper` is attached to the `front` with two M3x20 diagonal screws. In addition, the two M3x16 mating screws also hold the parts. 

Later, the extruder is attached to the `upper`, while the hotend is attached to the `front`. This means that any imperfections in the angle of the `upper` and `front` to each other make the extruder and hotend out-of-plane to each other. 

This warp does not become apparent until the fan assembly is fit, when force must be applied to re-align the extruder and hotend so that the fan should mates correctly and can be bolted.

The problem, I believe, is that the angle between the parts is affected by the diagonal and horizontal screws. Because the diagonal screws are at a 45° angle wrt to the horizontal screws, they pull the parts out of alignment.

After discovering this issue, I rechecked and filed the printed parts to ensure that they mated correctly. While this improved the collinearity of the extruder and hotend, it did not solve it completely. Also, it is nearly impossible to judge the alignment with just the parts.

The problem is visible in Kapman's Basement Workshop's [TAP video](https://youtu.be/5FaUZ4w1jmI?t=1070) at 7:50. When he tightens the lower screws, a gap appears at the top of the fan shroud. 

The result of the warp is that aligning the bolts for the fan shroud is made more difficult, as well as causing internal stresses in the system when it is forced to fit for final assembly.

Suggestions:
• Redesign the parts to correct the issue.
• Until then, add a note in the manual to reduce confusion when users notice the misalignment. How much of a gap between the extruder and the fan shroud is acceptable? How much force should be applied to make the extruder and fan connect?

Badnoob-327, 2023-01-25:
we are seeing some cases of this and were all the way back to beta. Thought we had fully addressed it, but I can see its still happening in some cases.  I'll take another look, thanks.

Redwish89, 2023-03-26:
Any updates on this? There seems to me more reports on the Voron Discord without a clear cause or solution to prevent/fix the gap.

leeeboo, 2023-07-03:
I have the same issue. I have made 3 taps and stealthburger, all have this issue.