VoronDesign/Voron-Tap #50: Cracks on top bolts on [a]tap_center_r6

SinisterRj, 2023-02-13:
Trying to debug some ressonance issues on my printer I was checking the overal boolts torques and accidentaly break this piece. But looking at it, I felt that this part is verry week at this point, with only 3 walls from here the bolts make contact do the end of the part. I was able to increase this number to 5 without reducing the clearences to the other parts. And in my modification I was abble to cincundate the entire rail carrige, also increasing the stiffness of the part a little bit. The new part is printed and it is working on my printer.

[image]
[image]
[image]
[image]
[image]

raymondh2, 2023-02-21:
Hi. Could you please provide what material and color the part was printed in?

SinisterRj, 2023-02-21:
It was printed in Dental White ABS MG94 from GTMax3D (a brasilian 3d printer manufacturer and one of the best filaments in Brazil)

nemgrea, 2023-03-14:
unfortunately white filaments are known to be less robust compared to their equivalent darker colored counterparts due to the colorants needed for getting white filament.

SinisterRj, 2023-03-14:
Yes, I know. The new part I printed on black. But, anyway, it gave me an opportunity to improve a weak point of the part.