VoronDesign/Voron-Tap #51: LDO MGN12H Carriage End Caps Not Flush - Possibly Others

Darkstar77, 2023-02-13:
My 2.4r2 is a LDO kit. The LDO kit MGN12H carriage has a ~0.55mm drop from the main carriage body to the end caps. It seems like a pretty small amount, however I've now had my belt slip and strip out the belt retainers on the TAP centre piece 3 times in one day. I was taking it apart to resolve a loose MGN9 rail issue, and had printed the R6 revision of the centre. 

I decided to remove the centre assembly to replace it with the newer revision. I noticed that the tooth immediately beside the hole for the belt to pass through had stripped off, with damage to it's neighbor. This was the upper belt, and only one side of the centre was effected. This centre had about 200 hours of printing on it. 

I reassembled it with a brand new centre, and on tensioning the belts, the upper belt again slipped. When I took it apart, the teeth were stripped in the exact same way. There appears to be no damage to the belt.

I printed a third centre, added some 1mm foam to the end caps of the LDO MGN12H carriage, and carefully re-assembled it. Tensioned the belts, and it looked good. Re-assembled the print head, and started a print. It got about 10 minutes into the print, and the belt slipped again. Same deal, the tooth immediately beside the pass through hole was stripped off for the upper belt on one side.

I saw at least one user with the same problem on Discord, with the same non-flush end cap on their carriage too. This was not an LDO carriage. When I inspected the carriages on my switchwire, I noted that the end caps were not flush on those carriages either. Those were generic china rails. I'd like to blame user error here, just say I was too rough, over tensioned the belts, etc.. This third centre I printed today, when I assembled it, I was very gentle with the assembly, and the tensioning. If anything, the belts were slightly under tensioned. 

0.55mm doesn't sound like a lot, but it is apparently enough to allow some wiggle room for the belt to move, and as a result it is really easy to strip the belt retention. Once the tooth closest the pass through hole is gone, it is no longer able to hold the belt, and it slips. That essentially means that a single tooth is properly engaging the belt in this scenario, and nothing else is really retaining the belts. This makes the whole thing very easy to strip out and fail. The LDO rails are widespread, and by no means the minority.

Darkstar77, 2023-02-13:
I designed and printed a tiny spacer for my carriage, which makes the endcaps flush with the rest of the carriage. So far, new centre is installed and tensioned, and nothing is looking off, no belts slipping. I believe this is definitely an issue in respect to the belt retention on the X carriage.

jjfs127, 2023-02-16:
Same LDO 2.4r4 kit with same rail and consequently same problem. My solution was to extend the ridges of the rail mount higher to apply more pressure to the belt against the rail bed. I also reported this problem in the Voron tap discord channel. I hope this issue gets more traction and an official fix is provided.

@Darkstar77 would you mind sharing the spacer?

Darkstar77, 2023-02-17:
> 

@jjfs127 I had been meaning to setup my personal repo, so here you go: https://github.com/Darkstar77/Voron/blob/main/2.4/stls/TAP-LDO_MGN12H_Spacer.stl

It's challenging to print. 0.1 on the first layer, and 0.1 layers. I had to give a little extra squish and width to get the first layer to hold together on the thin sections. The thin portion is really just to help align everything, once you tighten the TAP centre down, it's not going anywhere.

nedi89, 2023-02-21:
I‘m having the same problem. Vivedino carriage. I’ll try to see if it works with spaces but I hope voron team comes up with an official solution.

Darkstar77, 2023-02-22:
@nedi89 
> I‘m having the same problem. Vivedino carriage. I’ll try to see if it works with spacer but I hope voron team comes up with an official solution.

You might give that spacer I linked a go. Not sure it will fit that rail, but it's solved my issue for the time being. It's tricky to print, because it needs a 0.1 first layer, but otherwise is a 3 minute print.

I've been looking at the issue, and the only thing I can think of would make the centre thicker, which is not great. Essentially moving the holes for the belts to the inside of the 4 screws holding the centre to the carriage. This would clamp the belts firmly against the steel of the carriage, with clamping force more evenly distributed, much like the original carriage. The down side is clearance around the MGN9 rail and carriage, and why I believe it would add additional thickness to the centre and the overall TAP. Not sure folks want to loose another 2 ~ 3 mm on Y.

nedi89, 2023-02-22:
I went back to euclid and cw2, I'm going to wait this one out, hopefully Voron team comes up with a solution, don't want to buy a whole new rail because of the TAP.

Darkstar77, 2023-02-24:
Welp, after about 30 hours of printing with the spacer in place, I am back to square one, because my belts are slipping again. I got through a bunch of PLA prints, without issue, but when I went to do a long ABS print, chamber up to 55C, the belts slipped again. I haven't taken it apart, but I can guess what I will find, which will be teeth ripped off again. This needs to be addressed.

jjfs127, 2023-02-24:
@Darkstar77 You can give my mod a go. I haven't seen mine slip. I extended the ridges higher, now,  whether they apply too much pressure on the belts, I'm not sure.
[[a]Tap_Center_r6_modified_ridges.zip](https://github.com/VoronDesign/Voron-Tap/files/10828451/a.Tap_Center_r6_modified_ridges.zip)

nedi89, 2023-02-24:
Yeah, I tried your spacer first, it doesn't work for me. I like the idea of TAP but I personally think this mount design is not really thought through. There is just not enough force pushing those belts against carriage, the holding point is too far out on the edges, belts are going to slip sooner or later, or you have to keep them fairly loose. Best solution would be to place those ridhes closer to the middle, where you can really apply some force with those 4 screws but to be honest, I don't see how's that even possible.

Darkstar77, 2023-02-25:
@jjfs127 I tried your part, it failed on the first print, same deal. I see what you did though, and how you did it. Extruding the faces on top of the teeth. It's a good idea.

I've taken the CAD for the centre, and did some work on it. Similar to yours, but I took the time to re-sculpt the teeth to match their original profile, as well as increase the overall height of the surrounding shoulder that interfaces with the end caps. I am hoping that increasing the should by 0.2, and the teeth by 0.4, that will take up all the slack the end caps on the LDO carriage have, and will work. Printing it right now, and will be installing it later. If it works, I will upload it to my github, and link it.

jjfs127, 2023-02-25:
@Darkstar77 Sorry to hear it didn't work but I think you're on the right track. I thought of what you'd did alas my cad skills are novice at best. Let me know how it goes..

Darkstar77, 2023-02-25:
@jjfs127 here is the difference between the original R6, and what I've done.
https://github.com/Darkstar77/Voron/raw/main/TAP/OriginalCentreNewCentre.png

nedi89, 2023-02-25:
If it works out for you @Darkstar77 maybe I’ll give it another shot.

x-c-dog-x, 2023-03-02:
My first print of the Tap Center r6 had been running just fine for a few hundred hours in my LDO V2.4r2 build until I finally finished enclosing it. Within a few days the belts started to slip. Tore  everything apart and found out the same problem as above. Printed another center and it too failed. Tried the modified Center, but it too failed within a hour or so. Printed another and cut some plastic spacers out of some sheet plastic and stuck them on the rail carriage ends with double stick tape, just to hold them in place for installtion. Put things back together with a modified Center. This one lasted for a few more hours, but failed with Center cracking at the wings. All these Center were printed in Prusament ASA. Have now printed one in eSun ABS+, that seems a little more pliable. Time will tell. About ready to break out the CAD and redesign the whole belt retension method.

Darkstar77, 2023-03-06:
@x-c-dog-x I totally understand the frustration. I've got 2 versions I've modified, one of which is about 30 hours into printing now, and has been working. It is stressed a little, which is why I designed a small revision to that, which I've printed but not installed yet. If you're curious, you can check that out here: https://github.com/Darkstar77/Voron/tree/main/TAP I recommend the 6.2 part, which is the latest revision, and should help.

I've written up some suggestions on assembly, because I believe part of the issue is damaging one or more of the teeth on the center part during assembly. They are on the above github page, but it boils down to careful assembly around pre-tensioning the belts during assembly, to avoid damaging the teeth on the center, which is very easy to do.

I tried the "spacer" solution too, and it worked, sort of, until I printed ABS enclosed, and then it didn't. I wasn't able to get a spacer printed in ABS that would work sadly, and moved on from the idea. Metal shim material would probably be the best, but again, you have the potential for it to slip, or move. I can think of a simple design for an almost clip on piece of metal, but I don't have the tools to make it from the suitable stainless steel. It would need to be punched out, and bent on a break, neither of which I have access to. Could probably laser cut it, but I also don't have access to that.

x-c-dog-x, 2023-03-06:
Yeah, not really in favor of having the belts push down any more on the endcap wipers. Surely, will cause other issues. Still, at this point, I went back to using the stock Center and cut a couple strip spacers of a HDPE plastic lid and attached them with double-stick tape (just to hold them during assembly) on top of the endcaps. So far, after about 60 hours of enclosed printing they're hiolding up. Now I see another issue that will probably invole extrusion backers to help keep things in line after everthing heats up. WIll have to look at 6.2 when I get the chance.

nedi89, 2023-03-06:
I printed and installed the 6.2 centre , so far it's doing great after around 20 print hours. I'm printing only ABS so it should really put it to the test. I'll keep you updated.

nedi89, 2023-03-08:
Just an update, after around 40 print hours, my belts slipped again and I got a nasty layer shift. I dissasembled the whole thing and went back to old cw2 mount system with euclid probe and I think this time it's for good. I im no mood to assemble and disassemble every couple days.

Darkstar77, 2023-03-08:
@nedi89 were the teeth damaged where the belt slipped?

I'm beginning to suspect that the only way to really fix this is going to be to make it thicker by about ~3mm, so the slots can be moved inboard and come out by the rail. That way the clamping force would be more even, with a large engagement area. It would place the belts inboard of the screws... I'm going to look into that as an idea.

nedi89, 2023-03-08:
@Darkstar77  yes the teeth are damaged, broke off basically, the ones that didn't broke were deformed, was just waiting to happen. I should say that after initial belt tightening after assembly, I didn't try to retighten it or anything. Yeah, i also replaced my Vivedino X rail and carriage with a genuine HIWIN ones, because Kapman mentioned he didn't have any issue with the TAP on his HIWIN.

jjfs127, 2023-03-08:
I bit the bullet and bought another mgn12 rail. I went for a CNA rail as per the BOM. I will be going back to the standard piece and will try it in ASA-CF, hopefully with better results.

macmanluke, 2023-03-18:
Defintely had issues with this on my LDO rails (although less since i made sure to loops the belts fully around)
Just had the outer bits break off so will try this version.

sf-pfm265, 2023-04-19:
@Darkstar77 any update on your v2 design? I'm running an LDO Trident with Klicky PCB and I'm thinking of migrating to Tap.  I could switch out my rail to a different manufacturer as some here have done, but if your V2 fix is working, then I'll give it a try.

Darkstar77, 2023-04-28:
> @Darkstar77 any update on your v2 design? I'm running an LDO Trident with Klicky PCB and I'm thinking of migrating to Tap. I could switch out my rail to a different manufacturer as some here have done, but if your V2 fix is working, then I'll give it a try.

The latest version, which I've been running, is on the linked github. It's versioned 6.2, and so far it's what I've been running. I have to admit that I purchased a CNC TAP kit now, so I will be replacing it eventually. So far my version of the TAP center has been holding up well. That said, it still failed for another person in the same way. I have been tinkering with a mod that routes the belt path under the screw that secures it to the MGN12H carriage, to apply more clamping force, but so far my attempts at that add about 4mm of thickness to the part, and you lose ~4mm on Y. I'm not happy with that.

DrFreeman001, 2023-05-02:
Same problem here. LDO Kit.

macmanluke, 2023-05-02:
V8 is out now which fixes this problem (as does cnc tap)

Darkstar77, 2023-05-03:
> V8 is out now which fixes this problem (as does cnc tap)

Looked it over, and yes, that is similar to the direction I was going in to try to solve this. Pretty happy to see that. Updated my original post to direct people to the R8 version.

Badnoob-327, 2024-06-27:
Closing this as fixed with R8

closed 2024-06-27 by Badnoob-327