VoronDesign/Voron-Tap #59: Incorrect Part Placement?

thijstriemstra, 2023-04-21:
hi @tanaes I submitted the bom and placement files for an order at jlcpcb (my first) and uploaded the files provided in https://github.com/VoronDesign/Voron-Tap/tree/main/OptoTap/Tap_Photosensor_PCB_v2/jlcpcb

They came back with the following issue though, so it seems the PCB placement is off or I did something wrong.. anyway, I guess this corrected version is ok?

[image]

thijstriemstra, 2023-04-24:
> Since we are not so sure about the polarities of the components D1 are correct or not. Could you please kindly check if the polarities and placements of the components are correct in the below picture? Is it okay to proceed with production?

I opened the `Tap_Photosensor_PCB.kicad_pro` file in KiCad 7.0 and it shows the following schematic for D1:

[image]

[image]
 
I guess that'll be enough info for them to answer their question about D1, but I'll also include a [link to the kicad files](https://github.com/VoronDesign/Voron-Tap/tree/main/OptoTap/Tap_Photosensor_PCB_v2).

tanaes, 2023-04-24:
The positive terminal should be in the upper left of the footprint in that picture — their “corrected” part is upside down.

thijstriemstra, 2023-04-29:
thanks for the feedback @tanaes. they returned with:

[image]

tanaes, 2023-04-29:
That is incorrect -- D1 should be rotated 180°. Pin 1 should be in the
upper right corner, as denoted by the corner mark in the silkscreen.

thijstriemstra, 2023-05-04:
[image]

thanks @tanaes, they came back with this, and it looks ok now right?

tanaes, 2023-05-04:
That is correct!

—
<https://github.com/VoronDesign/Voron-Tap/issues/59#issuecomment-
auth/AB7ISADPAQJN5DDBKJCORGLXEONUNANCNFSM6AAAAAAXGGLK3I>.

thijstriemstra, 2023-05-05:
thanks for the help, the production started. how can we prevent this from happening to other users, it seems the jlcpcb .zip files in the repository produce incorrect jlcpcb instructions? can they updated? im not sure how because  i don't see an export to zip function in the jlcpcb interface..

tanaes, 2023-05-05:
initially I think JLCPCB had incorrect pin numbering for this part in
their system, so what would happen is the original automated placement
would be correct but then they would go back and manually 'fix' it to be
backwards. My guess is that they have updated their database but that
has broken the rotation for that part in the file. I will need to re-
export and check again.

—
<https://github.com/VoronDesign/Voron-Tap/issues/59#issuecomment-
auth/AB7ISAGNZ733DGNNNSPNBK3XERJ3ZANCNFSM6AAAAAAXGGLK3I>.

thijstriemstra, 2023-06-13:
looks like this was fixed in https://github.com/VoronDesign/Voron-Tap/commit/e783ce8710ebc401b4b872ef84dd3216f443cf81, thanks!

closed 2023-06-13 by thijstriemstra