VoronDesign/Voron-Tap #63: Continue to support MGN9H

jueank, 2023-05-03:
I would like to see the team continuing to support MGN9H rails (at least a single rail) instead of only MGN12H. There is still a fairly large group of Voron owners using MGN9H. Similarly to support a variety of toolheads (like still E3D V6 and more), MGN9H should not be forgotten. I don't see a technical reason to make such a hard cut but only because of not wanting to spend the effort.

tnibler, 2023-05-03:
> but only because of not wanting to spend the effort

Which is a totally valid reason of not doing work for free, is it not?

jueank, 2023-05-03:
Yes, perfectly valid. But I clearly vote for the continuation. In my opinion, this is an example of relativism in its purest form, because you could question any topic if you put the voluntary effort against it. Stop working on Vorons. it's work for free. You could stop it immediately. How would you want to argue against it? I don't think we should start a fundamental disussion about MGN9H vs MGN12H. The lead developers should just think about it and know that there are users interested in TAP but otherwise happy with MGN9H. Yes, I know, you can do mods on your own. But this is far different from being officially supported and this is exactly what I'm asking for here.

JanB97, 2023-05-03:
I am not as in to the current voron team, but could you not do a MGN9 mount and make a pull request? 
Maybe they will upstream your design.

Clearly the message is that they are not happy with the mgn9 design and don't want to support it so people get rid of it.

Badnoob-327, 2024-06-27:
I don't think it is possible to use MGN9H rails in tap, without compromising the design.  I did try early on (ad I also had a MGN9 dual rail).

closed 2024-06-27 by Badnoob-327