VoronDesign/Voron-Tap #71: Incorrect bolt length for rail

Furcube, 2023-05-26:
Manual specifies M3 6mm. 8mm and 12mm With part which have old version of [a]_Tap_Center_r8.stl.
8mm and 12mm bolts are too short for current version.

In step file bolt lengths are 6mm, 12mm and 16mm.

Martorias, 2023-07-20:
It's mentioned in the r8 errata; https://github.com/VoronDesign/Voron-Tap/blob/main/Manual/R8_errata.md
I didn't find this until I started building it as well so perhaps it's time to update the regular manual.

widarlein, 2023-09-01:
Yup, as @Martorias said. It's in the errata. There's no mention of the errata in the main readme though, just a link to the manual so I found it quite confusing. I ended up melting the heatset inserts farther into the plastic yesterday when the bolts didn't reach and I hadn't read the errata. 

I also missed the other steps in the update but those weren't irreversible so I could take it appart and fix.

The manual should probably be updated or there should be clearer links to the errata at least