VoronDesign/Voron-Tap #79: Magnet Holder possible improvement

xanxaz, 2023-09-05:
The Magnet holder screw position on the left side(front view), when you tighten it it will pull back the magnet holder and away from the set position(inward), making it a pain to set properly, maybe instead of a side screw, screw the part from the bottom? so it doesnt change the position/rotate the holder when tighten? the other side gets forced into the screw (outward) so it doesnt matter much.

igiannakas, 2024-04-05:
I agree with this - screwing the magnet from the side results in it twisting as the screw rotates. It makes it nearly impossible to retain the same position if you apply any meaningful torque to the screw. This would be a big QOL improvement.

Badnoob-327, 2024-06-27:
Sorry for such a late response, and thank you for your suggestion. 
The magnet holding screws should not be overtightened, it will lead to problems cracking the thin print sections.  If our magnets are in contact properly, tighten the screw with moderate torque only.

Screwing from the bottom won't work without a reliable locknut method, the toolhead shakes too much, and the screws will back out over time.

closed 2024-06-27 by Badnoob-327