VoronDesign/Voron-Tap #95: Attachment method (hex nut or heat-set) not specified for belt covers

paynterf, 2025-07-11:
[image]

The method for securing the M3x8 BHCS screws to the Tap_center part is not shown (there is a translucent drawing, but is that a hex nut or heat-set part).  The hole in the back of the Tap_center part appears to be sized for a heat-set screw, but there's not much plastic thickness.

kiwidave72, 2025-07-11:
It's a hex nut

paynterf, 2025-07-11:
Hmm, the actual piece doesn't have a hex-nut outline, so I'm not sure how it stays in place while the part is being attached to the rail carrier.  As I mentioned above, I wound up using two heat-set inserts, and they work - sort of.  I would print a new part and try again, but unfortunately my printer is down for Voron Tap installation ;)